jpcurti/ender3-v3-se-klipper-with-display #15: Make serial bridge to usart2 as default, hide other ports

jpcurti, 2024-05-31:
Currently we need to build this modified Klipper and configure which port we want to bridge via:

```sh
make menuconfig
# Choose option "USART2" and save
make
```

And also configure the serial bridge parameters in **printer.cfg**. Both steps are not necessary in this case as the serial port, baud, eol and other parameters are fixed for the ender3v3se.

It would be good to have this pre-configured, so that in the Serial bridge list from **make menuconfig** the USART2 is selected by default and no configuration on **printer.cfg** is necessary.

jpcurti, 2024-06-01:
Not sure if this should be done, as it will change the serial_bridge source files.

SupremeCake17, 2024-06-02:
yes, i think it would be better just to do that.

closed 2024-06-02 by SupremeCake17