jpcurti/ender3-v3-se-klipper-with-display #41: Klipper crasing when going into Move menu

pwlgrzs, 2024-06-02:
**Describe the bug**
Klipper crasing when going into Move menu

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Prepare'
2. Click on 'Move'
3. Klipper crashes.

**Expected behavior**
I should be able and change XY positions.

**Screenshots**
[image]

**Additional context**
klippy.log attached.
[klippy.log](https://github.com/user-attachments/files/15525727/klippy.log)

SupremeCake17, 2024-06-02:
from your log file, it looks like the position X is a negative number. Are you sure the printer is homed? What does the web UI say the position is?

pwlgrzs, 2024-06-02:
> from your log file, it looks like the position X is a negative number. Are you sure the printer is homed? What does the web UI say the position is?

It was happening after homing as well. Got fixed now.

SupremeCake17, 2024-06-03:
How did you get it fixed? Can we close this issue now?

jpcurti, 2024-06-04:
This was indeed a bug. The attributes that hold the current axis weren't fetched before going to the move menu, which causes this and other menus to crash klipper. This was patched on PR 40: https://github.com/jpcurti/ender3-v3-se-klipper-with-display/pull/40, therefore I am closing this issue.

closed 2024-06-04 by jpcurti