jpcurti/ender3-v3-se-klipper-with-display #42: Changing axis position with knob is slow

pwlgrzs, 2024-06-02:
**Describe the bug**
Changing axis position with knob is slow

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Prepare'
2. Click on 'Move'
3. Scroll down to any axis and try to change the value
4. Value changes slow and faster you turn the knob the slower it goes

**Expected behavior**
Faster knob turning = faster value change (as in stock)

jpcurti, 2024-06-04:
The current step for the axis move is 0.1mm. I believe that we could have it set to 0.5mm for the axis without any problems. Later on we can use the "fast turn" movements from the decoder module to make something more dynamic

broyuken, 2024-07-25:
I second this, just coming from Marlin it's great that the screen works again with klipper but the knob speed should be sped up if possible.

T9Air, 2024-08-05:
Why hasn't this been merged yet?

PR #107 (2025-02-20): feature:  Add support for fast_up and fast_down encoder events
This pull request to `klippy/extras/e3v3se_display.py` includes changes to support fast encoder rotations and improve the handling of key events.

Fixes #42 

Enhancements to encoder handling:

* Added new encoder states for fast clockwise and counterclockwise rotations (`ENCODER_DIFF_FAST_CW` and `ENCODER_DIFF_FAST_CCW`) and updated the existing states to accommodate these changes.
* Updated the `key_event` method to handle fast rotation events (`fast_up` and `fast_down`).
* Added the `get_encoder_multiplier` method to determine t [... cut]
closed 2025-02-20 by SupremeCake17