jpcurti/ender3-v3-se-klipper-with-display #52: print queue

ArgatuIoan, 2024-06-03:
**unprecise print queue**
in the print menu of the printer, the files that are shown are the ones that i have printed *before* flashing the printer

**updating the queue**
would be amazing if the print que thingy would update when i print/upload new things to klipper

jpcurti, 2024-06-04:
Hi. Thanks for your feedback.

Are you sure it isn't just because the file list on Printing Menu is alphabetically ordered? Could you provide some images as example so we can take a look on that?

Have a nice week

ArgatuIoan, 2024-06-04:
Oops, I think ypu are correct (might be placebo, but now i think i remember it being the same on marlin)
[image]

jpcurti, 2024-06-05:
I would keep the current file list (alphabetical order) atm and maybe we can revisit this topic after we implement the subfolder feature