jpcurti/ender3-v3-se-klipper-with-display #70: screen not working

T9Air, 2024-07-02:
Hi I used your .bin to update my printer, but when I did, the screen still did not work.

SupremeCake17, 2024-07-02:
Did you add it to your config? it wont work without that. also make sure you actually changed your klipper repo. and make sure the name of the bin file is different from the one you previously flashed. there are more detailed instructions here: https://schnoog.eu/hobbies/3dprinting/ender-3-v3-se-update-to-make-the-display-working

T9Air, 2024-07-02:
I did.

SupremeCake17, 2024-07-02:
Your Klipper Installation is this and not the official one, right? You deleted your old one and installed this one? Or changed it in kiauh and reinstalled Klipper?

T9Air, 2024-07-02:
Never mind, I redid it and it worked. Sorry for the bother.

closed 2024-07-11 by SupremeCake17