jpcurti/ender3-v3-se-klipper-with-display #76: mcu 'mcu': Unknown command: serial_bridge_send

T9Air, 2024-08-05:
I just updated to the newest version of Klipper, and if I put in 
```
[e3v3se_display]

language: english
```
I get this error: mcu 'mcu': Unknown command: serial_bridge_send

rojekj, 2024-09-05:
Same problem here. Tried with my own compilation as well as the provided bin file.

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: 1.0.0-146-g60b65f82
MCU(s) which should be updated:
<none>
Up-to-date MCU(s):
mcu: Current version 1.0.0-146-g60b65f82

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'mcu': Unknown command: serial_bridge_send

T9Air, 2024-09-05:
@rojekj I was able to figure out the problem recently. Did you enable the serial bridge for usart2 when making your own config? I am pretty sure that that fixed the problem.

rojekj, 2024-09-05:
@T9Air Thank you for the tip, that did the trick!

closed 2024-09-06 by T9Air