jpcurti/ender3-v3-se-klipper-with-display #81: Error on section

digitalinima, 2024-08-20:
When i add [e3v3se_display] My printer.cfg, an error to boot It happens and my device won't start.

It says unrecognized section. I use Mainsail

Do I need to use fluidd??

kcajjones, 2024-08-21:
I get the same error with Fluidd. Any help would be greatly appreciated.

igoralexeev, 2024-08-21:
same problem

kirmiziben10, 2024-08-24:
**MAKE SURE TO TAKE BACKUP OF YOUR ORIGINAL KLIPPER FOLDER.**
You need to use this repo as your new klipper. Just stop klipper from the web interface and replace this repo as your klipper folder. All my previous configs worked after just adding this to the printer.cfg

```
[e3v3se_display]
language: english
logging: True
```

SupremeCake17, 2024-08-25:
If you are using KIAUH, you need to switch the repo there and reinstall it, then reflash klipper. There is a guide here: https://schnoog.eu/hobbies/3dprinting/ender-3-v3-se-update-to-make-the-display-working
If you aren’t, just delete the current Klipper (~/klipper) and clone this one. Then build and flash, just like the Klipper install instructions.

igoralexeev, 2024-08-28:
> @digitalinima @kcajjones @igoralexeev **ОБЯЗАТЕЛЬНО СДЕЛАЙТЕ РЕЗЕРВНУЮ КОПИЮ ОРИГИНАЛЬНОЙ ПАПКИ KLIPPER ПЕРЕД ТЕМ, КАК ЧТО-ТО ДЕЛАТЬ.** Вам нужно использовать этот репозиторий в качестве нового klipper. Просто остановите klipper из веб-интерфейса и замените этот репозиторий в качестве папки klipper. Все мои предыдущие конфигурации работали после простого добавления этого в printer.cfg
> 
> ```
> [e3v3se_display]
> language: english
> logging: True
> ```

all this has been done, still a mistake. The last time I tried to install it and it generally gave me an error when downloading git

destaben, 2025-08-10:
You need to use this version of klipper in the host. It will solve the issue