jpcurti/ender3-v3-se-klipper-with-display #90: Printer shuts down on starting print with [e3v3se_display] in config

valikvicious, 2024-12-01:
**Describe the bug**

I have two Ender 3 V3 SE printers connected to the same OrangePi 3 Zero. I've replaced Klipper to the one from this repository, flashed the .bin to both printers, and while idle, the display is working normally. 
However, when I start a print task while [e3v3se_display] section is present in the printer.cfg file, the printer stops with the following error:

```
MCU 'mcu' shutdown: Missed scheduling of next digital out event
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Fluidd GUI
2. Select one of the printers 
3. Rightclick on the uploaded gcode in "Jobs"
4. Choose print
5. See error

**Expected behavior**
Printing starts as normal

**Relevant klippy.log piece**
`
Received 3572.118963: b'{"id": 281473551949584, "method": "gcode/script", "params": {"script": "SDCARD_PRINT_FILE FILENAME=\\"\\u041a\\u043e\\u0440\\u043e\\u043a_\\u0441\\u0442\\u0430\\u043d\\u0434\\u0430\\u0440\\u0442\\u043d\\u0438\\u0439_\\u043f\\u0456\\u0434_\\u0446\\u0432\\u044f\\u0445_2,45\\u043c\\u043c_PETG_52m22s.gcode\\""}}'
mcu 'mcu': Exception in serial callback
Traceback (most recent call last):
  File "/home/klipper/klipper/klippy/serialhdl.py", line 56, in _bg_thread
    hdl(params)
  File "/home/klipper/klipper/klippy/mcu.py", line 641, in _handle_shutdown
    self._serial.dump_debug())
  File "/home/klipper/klipper/klippy/serialhdl.py", line 280, in dump_debug
    cmds = self.msgparser.dump(msg.msg[0:msg.len])
  File "/home/klipper/klipper/klippy/msgproto.py", line 270, in dump
    params, pos = mid.parse(s, pos)
  File "/home/klipper/klipper/klippy/msgproto.py", line 185, in parse
    v, pos = t.parse(s, pos)
  File "/home/klipper/klipper/klippy/msgproto.py", line 49, in parse
    c = s[pos]
IndexError: index too large for cdata 'uint8_t[]' (expected 5 < 5)
Virtual sdcard (1697): '113.524],[127.294,113.619],[126.118,117.237],[126.086,117.331],[125.866,117.796],[125.633,118.254],[125.386,118.705],[125.329,118.786],[123.093,121.864],[123.033,121.943],[122.68,122.317],[122.317,122.68],[121.943,123.033],[121.864,123.093],[118.786,125.329],[118.705,125.386],[118.254,125.633],[117.796,125.866],[117.331,126.086],[117.237,126.118],[113.619,127.294],[113.524,127.323],[113.019,127.418],[112.511,127.498],[112.001,127.564],[111.902,127.565],[108.098,127.565],[107.999,127.564],[107.489,127.498],[106.981,127.418],[106.476,127.323],[106.381,127.294],[102.763,126.118],[102.669,126.086],[102.204,125.866],[101.746,125.633],[101.295,125.386],[101.214,125.329],[98.1362,123.093],[98.0569,123.033],[97.683,122.68],[97.3195,122.317],[96.9668,121.943],[96.9074,121.864],[94.671,118.786],[94.6139,118.705],[94.3674,118.254],[94.134,117.796],[93.914,117.331],[93.8821,117.237],[92.7064,113.619],[92.6771,113.524],[92.5819,113.019],[92.5015,112.511],[92.4359,112.001],[92.4345,111.902],[92.4345,108.098]]\n;TYPE:Custom\n'
Upcoming (2721): 'start_print BED_TEMP=70 EXTRUDER_TEMP=240 CHAMBER_TEMP=0\nG90\nG21\nM83 ; use relative distances for extrusion\n; filament start gco'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[337.0, 0.0, 10.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.016666666666666666 extrude_factor=1.0 speed=25.0
Reactor garbage collection: (3543.527610772, 0.0, 0.0)
MCU 'mcu' shutdown: Missed scheduling of next digital out event
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Dumping stepper 'stepper_z' (mcu) 2 queue_step:
queue_step 0: t=2551389615 p=1940 i=36180 c=995 a=0
queue_step 1: t=2587388715 p=2935 i=36180 c=995 a=0
Dumping trapq 'toolhead' 1 moves:
move 0: pt=34.465877 mt=1.990000 sv=5.000000 a=0.000000 sp=(0.000000,0.000000,0.025000) ar=(0.000000,0.000000,1.000000)
Requested toolhead position at shutdown time 35.838002: (0.0, 0.0, 6.885621319444427)
`

**Desktop (please complete the following information):**
 - OS: Windows 11 or ArchLinux
 - Browser: Chrome or FireFox
 - OrangePi OS: Orange Pi 1.0.2 Bullseye
 - Klipper: 1.0.0-218 
 - Moonraker: v0.9.3-3
 - Fluidd: v1.31.0
 - PrettyGCode: v0.12.0-2
 - Mobileraker: v0.4.0-88
 - KIAUH script version: v5.1.2

**Additional context**
The same behavior is observed with both self-compiled .bin as well as the one downloaded from release.
Klipper is the correct one, cloned from this repo.
The displays on both printers are working normally when the printers are idle.
Also, once the [e3v3se_display] section is removed or commented out in the config, the printer behaves normally, though the display is inert (showing the expected menu, but not being interactive).

Are the two printers bothering each other? The printer.cfg files are attached.
[printer1.txt](https://github.com/user-attachments/files/17967538/printer1.txt)
[printer2.txt](https://github.com/user-attachments/files/17967539/printer2.txt)

valikvicious, 2024-12-01:
Figured it out, the cause was the name length of the gcode file. Please feel free to close this.

closed 2025-01-14 by SupremeCake17

Andrew-Bulova, 2025-05-31:
Just for those who encountered a similar issue: I had the same error caused by Cyrillic characters in the filename. Renaming the file solved the problem.