jpcurti/ender3-v3-se-klipper-with-display #92: Latest version firmware doesn't work on printer.

kcajjones, 2024-12-17:
I just updated Klipper from this repository via the Fluidd web interface and then SSH'd into the raspberry pi to compile the 3d printer firmware. All went as expected. I moved the klipper.bin file to the Klipper config folder to allow me to download it vai the Fluidd web interface and it is 39kb in size as expected. After putting the klipper.bin file onto an SD card and allowing the printer to flash the firmware, it no longer communicates with Klipper/Fluidd. Restoring the previous Klipper firmware works fine.

leonardola, 2024-12-20:
I tried building the firmware from scratch with the latest version and my printer is bricked after flashing it. I tried to flash other firmwares but it simply does not work. Any ideas on how to fix that?

leonardola, 2024-12-20:
Never mind. I was able to reinstall the oficial firmware back.

kcajjones, 2024-12-22:
You can either use an older version of this klipper repo or put the official firmware back on.

SupremeCake17, 2025-01-14:
It seems to still work on my printer just fine. Could you download your klippy.log file and upload it here? I can take a look at it to see what is causing it to not connect.

Atomique13, 2025-01-16:
we got a user to reflash their printer with the latest version and it worked.

ArgatuIoan, 2025-01-16:
i can confirm it works just as before :)

closed 2025-02-18 by SupremeCake17