jpcurti/ender3-v3-se-klipper-with-display #95: menu in chinese insterad of any other language

aardjoun, 2025-01-01:
**Describe the bug**
I followed the blog post to implement the klipper firmware with the scrren and added the section in my printer.cfg
The screen work fine but the only language shown is chinese.

i try diferent languages (english, french, no option) but it still in chinese.

SupremeCake17, 2025-02-18:
Could you paste that part of your printer.cfg here, so I can make sure you have done it correctly? The only thing i can think of is that you have put it in your config wrong because it works for me and most other people using it. Also is your display firmware on 1.0.6? If it is still on 1.0.4, that could be the issue.

aardjoun, 2025-02-18:
by seeing your reply a try again to recompile the firmware but still the same issue.
I tried different languages and it workish. the languages doesn't match with the corresponding output :
English to chinese
Spanish to turkish (i guess)
german to english 
Portuguese to spanish
and so on.

here my printer.cfg config
`[include fluidd.cfg]

[include mainsail.cfg]

[include macro.cfg]

[include timelapse.cfg]

[include KAMP_Settings.cfg]

[firmware_retraction]

[exclude_object]

[e3v3se_display]
language: german 
# set german to show english

[virtual_sdcard]
path: ~/printer_data/gcodes
on_error_gcode: CANCEL_PRINT`