jpcurti/ender3-v3-se-klipper-with-display #98: Klipper crashes when viewing the 'Print' menu if there is no gcode available to print

Kamaroth92, 2025-01-21:
**Describe the bug**
When opening the "Print" menu via the screen Klipper will crash if there are no files available. On the printer the screen displays an expected empty list but the usual "Print/Back" labels are not there and moving/clicking the nob does nothing. 

Once there is a printable file on the printer it works as expected however if you then delete all of the gcode the issue re-surfaces.

From the Mainsail interface I get the error
`Unhandled exception during run
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown`

**To Reproduce**
Steps to reproduce the behavior:
1. Enter the print menu on the printer when there is no gcode on the printer. Most likely only an issue on a fresh install.

**Expected behavior**
The screen shows an empty list with Klipper does not crash.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: RaspberryPi 3
 - Browser: Brave

**Additional context**
Klipper: v0.12.0-410-gcf3b0475 (Build on the RasPi and flashed onto the printer; not using the release build)
Mainsail: v2.13.2

SupremeCake17, 2025-01-29:
Could you provide a klipper.log file from right after the crash happens? That will show the exact error that is happening so I can fix it easier.

atraides, 2025-04-12:
Hello,

I'm not sure if this is still needed or not, but I managed to reproduce the issue on a fresh Klipper install. Here are the logs from when it happened:
```
Stats 1638.6: gcodein=0  mcu: mcu_awake=0.003 mcu_task_avg=0.000013 mcu_task_stddev=0.000007 bytes_write=82427 bytes_read=120361 bytes_retransmit=0 bytes_invalid=0 send_seq=4548 receive_seq=4548 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=71998825  mcu_temp: temp=31.1 raspberry_pi: temp=44.8 heater_bed: target=0 temp=45.2 pwm=0.000 sysload=0.22 cputime=21.296 memavail=7640620 print_time=117.120 buffer_time=0.000 print_stall=0 extruder: target=0 temp=86.3 pwm=0.000
Unhandled exception during run
Traceback (most recent call last):
  File "/srv/klipper/klipper/klippy/klippy.py", line 176, in run
    self.reactor.run()
  File "/srv/klipper/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/srv/klipper/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/klipper/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
                            ^^^^^^^^^^^^^^^^^^^^^
  File "/srv/klipper/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/klipper/klipper/klippy/extras/buttons.py", line 82, in <lambda>
    (lambda et, c=callback, bt=btime, s=state: c(bt, s)))
                                               ^^^^^^^^
  File "/srv/klipper/klipper/klippy/extras/display/menu_keys.py", line 95, in click_callback
    self.callback('click', eventtime)
  File "/srv/klipper/klipper/klippy/extras/e3v3se_display.py", line 624, in key_event
    self.encoder_has_data()
  File "/srv/klipper/klipper/klippy/extras/e3v3se_display.py", line 4453, in encoder_has_data
    self.HMI_MainMenu()
  File "/srv/klipper/klipper/klippy/extras/e3v3se_display.py", line 749, in HMI_MainMenu
    self.Draw_Print_File_Menu()
  File "/srv/klipper/klipper/klippy/extras/e3v3se_display.py", line 3061, in Draw_Print_File_Menu
    self.Redraw_SD_List()
  File "/srv/klipper/klipper/klippy/extras/e3v3se_display.py", line 4137, in Redraw_SD_List
    self.lcd.draw_rectangle(
TypeError: TJC3224_LCD.draw_rectangle() takes 7 positional arguments but 8 were given
Transition to shutdown state: Unhandled exception during run
Dumping gcode input 0 blocks
Dumping 20 requests for client 548223932048
```

```
Klipper:    1.0.0-418
Moonraker:  v0.9.3-72

Fluidd:     v1.34.1
```