jpcurti/ender3-v3-se-klipper-with-display #104: Z-Offset adjustment menu

LeandroSQ, 2025-02-17:
**Is your feature request related to a problem? Please describe.**
When using the Z-Offset adjustment on Klipper, a menu will popup where you can move up and down the Z axis in order to calibrate your Z-Offset properly. Currently the screen does not support this custom menu.

**Describe the solution you'd like**
Would it be possible to create a custom menu? I can imagine something very simple like:
- Encoder rotation moves up and down
- Encoder press would toggle between different step sizes, 0.1, 0.5, 1.0 etc
- Double press could confirm the change

**Additional context**
I can help develop this feature, I am just not sure on whether the menus and logic we have on the screen are fixed and controlled by the screen firmware, or if we can display custom messages using Klipper directly. If this is possible, please, point me at the right direction and I can play with it to get some prototype in place.

SupremeCake17, 2025-02-18:
Yes, this is possible to do directly from klipper. The display firmware mainly just stores things like the images and icons, and klipper is what tells the screen to display those and where, and klipper can also display any text on the display, draw lines, and a few other things. All of the functions for the display are in TJC3224.py like drawing an image stored on the firmware or drawing text, and printerInterface.py has the functions for the screen to communicate with klipper (sending gcode commands to klipper) and for getting data from the printer. And then e3v3se_display.py handles everything else, it decides what to display and uses the functions in TJC3224 to display it, handles the encoder movement, and uses the printerInterface functions to control the printer. All of these files are located in /klippy/extras. Most of the messages are stored as images in the stock display firmware, but custom menus are still possible by drawing boxes and text (just like the feature not available popup). You can start by looking around in those files and seeing how everything works, and then have a look at the print progress menu. That menu automatically opens when a print starts, so you can look at how that works and use that to start on the Z offset menu. This sounds like a great feature for the display, you can start working on this and let me know if you need any more help with this!

LeandroSQ, 2025-02-18:
Thanks for pointing me at the right direction,
After tinkering a bit I got the initial part done: 

[image]

I'm able to detect when the manual probe calibration is running and show a popup, the number on the screen is the initial offset.
I just need to figure it out how to interact with the encoder and how to do a nicer UI.

LeandroSQ, 2025-02-18:
I figured it out how to interact with the encoder yesterday, looks like everything (other than the UI which is currently awful) is working.
The only question I have is how to detect a long press or double press on the encoder, do we have something like this currently?

SupremeCake17, 2025-02-18:
We do not currently have anything like that yet to detect double clicks, however, the time library is already imported at the top. you could get the time of when it was clicked and store that to a variable, and when it gets clicked again, compare the current time to the time of when it was last clicked, and if the difference is below the threshold time to register it as a double click, handle it as a double click, otherwise handle it as a single click. You can do it like this in the part where a click is detected:
```
if time.time() - self.last_click_time <= <double click threshold time>:
   <handle double click>
else:
   <handle single click>
self.last_click_time = time.time()
```

You compare the current time to the time of the last click, and see if it was below the time for a double click. After that, set the last click time to the current time. If you know of a better way, you can also do that if you want.

LeandroSQ, 2025-02-18:
Thanks for the pointer, again :)

Well, for detecting double or long presses I would need to debounce the encoder and delay it for some ms
I also saw some stuff on the MenuKeys that seem to have some logic for this, I wonder if we can reuse it on the `e3v3sedisplay.py`

LeandroSQ, 2025-02-18:
https://github.com/jpcurti/ender3-v3-se-klipper-with-display/blob/4983432ad2551fc2ff3b27286f3f0ace6cf8a7cf/klippy/extras/e3v3se_display.py#L556C4-L566C1

Might be able to set a flag on this keyEvent, we could introduce a new type but I'm afraid that would not be retro-compatible with existing menus.

SupremeCake17, 2025-02-18:
Long presses are already handled in MenuKeys, but in the KeyEvent function it sets both short and long presses to the same thing. You could try to make a new encoder state under long_click. It shouldn't mess anything else up, I don't think anything else uses long clicks. I would try that and test if everything else still works.

LeandroSQ, 2025-02-19:
@SupremeCake17 please take a look #105

closed 2025-02-19 by SupremeCake17