jpcurti/ender3-v3-se-klipper-with-display #119: error "Section 'e3v3se_display' is not a valid config section"

L0RD4N, 2025-03-04:
Hi, I flashed my SE printer and i keep getting the error "error "Section 'e3v3se_display' is not a valid config section"
, i tried compiling myself and the already compiled bin and nothing changes.
Cab anyone help? Ty

Thrilleratplay, 2025-03-05:
I had the same issue and could not figure what was I was doing wrong.  I [followed these instructions](https://schnoog.eu/hobbies/3dprinting/ender-3-v3-se-klippered) and it worked.  I still do not know what my original issue was.

ForbiddenEra, 2025-08-19:
You can't just use the firmware compiled from this repo, you also need to actually run the klipper from this repo on your host, if you use KIAUH or something, you can use advanced config to add this repo as your klipper repo.