jpcurti/ender3-v3-se-klipper-with-display #134: How to update Klipper version?

stuartparry2, 2025-06-09:
What is the recommended process for upgrading the Klipper version?

In the GUI, the option for Klipper is shown as "Invalid", due to "Unofficial remote URL".

Running the update option in KIUAH gives the following message:

>   fatal: bad object refs/remotes/origin/master
>   error: https://github.com/jpcurti/ender3-v3-se-klipper-with-display did not send all necessary objects
>   
>   warning: ignoring broken ref refs/remotes/origin/master
>   fatal: Not a valid object name origin/master

Looking through the various guides and posts, there are no references for specific update steps.

Please advise.

sp2917, 2025-06-09:
Self-resolved via an unintall/install using KIUAH. 

Config folder backups was run, but not required. 

No reflash of printer (klipper.bin) was required either.

sp2917, 2025-09-02:
@Sboshoff76 when I ran the uninstall/install process, I kept the Klipper repo to the jpcurtil one. This pulled the lastest klipper version as per the jpcurtil repo. I didn't try pointing the repo back to the offical Klipper repo.

I can now update Klipper via the Mainsail GUI, and it continues to reference the jpcurtil repo. I am assuming this is the intended method, i.e. via the GUI. 

I am assuming that I must have let my Klipper version get too far out of date and therefore the GUI/KIUAH could not perform the update.